kwindrem/RpiGpioSetup #3: GracefulShutdown uses gpioPin=16 also assigned to Relay 5?

JTG-IX, 2021-11-16:
First of all, many thanks for providing these setup helps! 
I was already very happy with the GracefulShutdown in the Venus GUI but now discovered in your code GracefulShutdown.sh that when gpio pin16 is pulled low the same could be triggered. (and a hardwire button makes sense in my case).
This pin however seems (?double) used to control relay 5 already in the gpio_list which may be the reason I cannot get it to work.

I tried modifying the gpio_list and can get gpioPin16 to change from out to in by commenting out relay 5 and adding a 6th input under the digital inputs. After reboot it is also shown as 'in' in  /sys/class/gpio/gpio23/direction. It does not seem to work though, when I take gpiopin16 (GPIO23) to ground nothing happens where I expect a gracefulshutdown.

Shutting down via the Venus gui works.

kwindrem, 2021-11-17:
the shutdown code I had in v3.1 and removed in v3.2 uses GPIO 16 (pin 36) for the input. Relay 5 uses GPIO 23 (pin 16) so there isn't a double use. Pins and GPIO numbers are confusing and I did not help by using the variable "gpioPin" !!!!!

In GracefulShutdown.sh it's really the GPIO NUMBER.

You should not need to add to or change the Venus gpio_list. In fact I chose a GPIO that is NOT used by Venus.

Hope that helps.

JTG-IX, 2021-11-17:
Thanks for your quick reply Kevin, appreciated.
Any reason why you removed it in v3.2?

kwindrem, 2021-11-17:
I removed it because I thought the GUI based shutdown was a better solution that didn't require hardware (a jumper wire). I also didn't want an input pin that could accidentally shut down the PI.

I could add the input back in with an option in the setup menu to install it or not. Comments?

JTG-IX, 2021-11-17:
I would certainly support having that option in addition so +1 from my side