kwindrem/RpiGpioSetup #5: Only 2 relays in 2.90~10

JimmyM01, 2022-06-03:
I've installed Venus OS (venus-image-raspberrypi4-20220519162847-v2.90_10.rootfs.img) from [johnny-brusevold
https://drive.google.com/drive/folders/18FYddU3oFxMUi_cNyQdWnQxRP1vMJNXm. After installing the RpiGpioSetup and SetupHelper there are only 2 relays present in the UI.
RpiGpioSetup version 3.7, SetupHelper version 4.13

kwindrem, 2022-06-05:
RpiGpioSetup replaces /etc/venus/gpio_list with one found in /data/setupOptions/RpiGpioSetup, or if that's not present a default in /data/RpiGpioSetup. The first one allows users to customize the relay mapping. The default should activate 6 relays.

JimmyM01, 2022-06-06:
After a reboot or 2 the 6 relays now appears in both the menus and the main page after installing the GUI enhancements.
Sorry to bother.