kwindrem/RpiGpioSetup #6: NodeRed Victron Relay just shows 4 relay

dersch81, 2022-10-02:
Hi, i'm just wondering how to get all 6 relay into the nodered module? Any idea?

[image]

kwindrem, 2022-10-02:
You should see entries for all relays defined in gpio_list in the Relay menu. Further, you should see parameters for each relay state (/Relay/0/State, etc) in com.victroneneryg.system in dbus-spy. If one or more are missing, I'd check the definitions in gpio_list. A default is provided in RpiGpioSetup which does activate all 6 relays. GPIO conflicts with other devices (hats for example) that might prevent them missing relays or digital inputs.

Beyond that, it would be something in Node Red which I know nothing about. Sorry.

pdcurtis, 2022-10-29:
@dersch81 You can also run `gpioinfo` which will show information on how all the GPIO lines are configured - it is best to limit the output somewhat by using `gpioinfo | grep GPIO` 

I use Node-RED for relay control and I know of nothing in Node-RED which should be causing your problem.

dersch81, 2022-10-29:
@pdcurtis thanks. are you able to use all 6 relays in NodeRed? 

My output of gpioinfo:

```
gpiochip0 - 54 lines:
        line   0:      unnamed       unused   input  active-high
        line   1:      unnamed       unused   input  active-high
        line   2:      unnamed      "sysfs"  output  active-high [used]
        line   3:      unnamed      "sysfs"   input  active-high [used]
        line   4:      unnamed      "sysfs"  output  active-high [used]
        line   5:      unnamed      "sysfs"   input  active-high [used]
        line   6:      unnamed      "sysfs"   input  active-high [used]
        line   7:      unnamed   "spi0 CS1"  output   active-low [used]
        line   8:      unnamed   "spi0 CS0"  output   active-low [used]
        line   9:      unnamed       unused   input  active-high
        line  10:      unnamed       unused   input  active-high
        line  11:      unnamed       unused   input  active-high
        line  12:      unnamed      "sysfs"  output  active-high [used]
        line  13:      unnamed      "sysfs"   input  active-high [used]
        line  14:      unnamed       unused   input  active-high
        line  15:      unnamed       unused   input  active-high
        line  16:      unnamed      "sysfs"   input  active-high [used]
        line  17:      unnamed      "sysfs"  output  active-high [used]
        line  18:      unnamed       unused   input  active-high
        line  19:      unnamed       unused   input  active-high
        line  20:      unnamed       unused   input  active-high
        line  21:      unnamed       unused   input  active-high
        line  22:      unnamed      "sysfs"  output  active-high [used]
        line  23:      unnamed       unused   input  active-high
        line  24:      unnamed       unused   input  active-high
        line  25:      unnamed  "interrupt"   input  active-high [used]
        line  26:      unnamed      "sysfs"   input  active-high [used]
        line  27:      unnamed      "sysfs"  output  active-high [used]
        line  28:      unnamed       unused   input  active-high
        line  29:      unnamed       "led0"  output  active-high [used]
        line  30:      unnamed       unused   input  active-high
        line  31:      unnamed       unused   input  active-high
        line  32:      unnamed       unused   input  active-high
        line  33:      unnamed       unused   input  active-high
        line  34:      unnamed       unused   input  active-high
        line  35:      unnamed       unused   input  active-high
        line  36:      unnamed       unused   input  active-high
        line  37:      unnamed       unused   input  active-high
        line  38:      unnamed       unused   input  active-high
        line  39:      unnamed       unused   input  active-high
        line  40:      unnamed       unused   input  active-high
        line  41:      unnamed       unused   input  active-high
        line  42:      unnamed       unused   input  active-high
        line  43:      unnamed       unused   input  active-high
        line  44:      unnamed       unused   input  active-high
        line  45:      unnamed       unused   input  active-high
        line  46:      unnamed       unused   input  active-high
        line  47:      unnamed       unused  output  active-high
        line  48:      unnamed       unused   input  active-high
        line  49:      unnamed       unused   input  active-high
        line  50:      unnamed       unused   input  active-high
        line  51:      unnamed       unused   input  active-high
        line  52:      unnamed       unused   input  active-high
        line  53:      unnamed       unused   input  active-high
gpiochip1 - 8 lines:
        line   0:      unnamed       unused  output  active-high
        line   1:      unnamed       unused  output  active-high
        line   2:      unnamed       "led1"  output   active-low [used]
        line   3:      unnamed       unused  output  active-high
        line   4:      unnamed       unused   input  active-high
        line   5:      unnamed "cam1_regulator" output active-high [used]
        line   6:      unnamed       unused  output  active-high
        line   7:      unnamed       unused   input  active-high

```

kwindrem, 2022-10-29:
It doesn't look like any of the GPIO pins that Venus OS typically uses are defined. Are you populating /etc/venus/gpio_list?

RpiGpioSetup will do this but you can also fill in the desired GPIOs manually. Refer to the gpio_list file included in RpiGpioSetup for clues.

pdcurtis, 2022-10-30:
@dersch81 @kwindrem Up to now I have always set my relays up directly rather than used RpiGpioSetup and have had access to as many relays as I required directly using the sysfs mechanism for any extras over the 2 supported by Venus. 

I have only recently installed RpiGpioSetup and I have checked and all 6 work from the Remote Console, they show up in dbus-spy and are correct in gpioinfo using defaults gpio-list.

```
root@raspberrypi4:~# gpioinfo | grep GPIO
	line   4:  "GPIO_GCLK"       unused   input  active-high 
	line   5:      "GPIO5"      "sysfs"   input  active-high [used]
	line   6:      "GPIO6"      "sysfs"   input  active-high [used]
	line  12:     "GPIO12"       unused   input  active-high 
	line  13:     "GPIO13"      "sysfs"   input  active-high [used]
	line  16:     "GPIO16"      "sysfs"   input  active-high [used]
	line  17:     "GPIO17"      "sysfs"  output  active-high [used]
	line  18:     "GPIO18"       unused   input  active-high 
	line  19:     "GPIO19"      "sysfs"   input  active-high [used]
	line  20:     "GPIO20"       unused   input  active-high 
	line  21:     "GPIO21"      "sysfs"  output  active-high [used]
	line  22:     "GPIO22"      "sysfs"  output  active-high [used]
	line  23:     "GPIO23"      "sysfs"  output  active-high [used]
	line  24:     "GPIO24"      "sysfs"  output  active-high [used]
	line  25:     "GPIO25"       unused   input  active-high 
	line  26:     "GPIO26"      "sysfs"   input  active-high [used]
	line  27:     "GPIO27"      "sysfs"  output  active-high [used]
	line   5:   "CAM_GPIO" "cam1_regulator" output active-high [used]
```

But @dersch81 is correct that only the first 4 are shown in the Victron Input and Output Nodes so it looks as if it is a restriction in the Relay input and output Nodes which are part of @victronenergy/node-red-contrib-victron . This could be a numeric limit or might be a block on certain GPIOs - unlikely but I will check.

IF you MUST have more than 4 relays extra ones are relatively easy to implement using the sysfs pseudo file system from an Exec node but since 2.90 you need to set them up in a startup script in rc.local as Node-RED no longer runs as root. They will not be accessible from the Remote Console or show on dbus unlike use of RpiGpioSetup which is a much more elegant solution but I have been using "sysfs" for some time and still am on my operational system on my narrowboat. I am happy to explain further.

pdcurtis, 2022-10-30:
@kwindrem 

Re
```
Custom GPIO list:
If /data/setupOptions/RpiGpioSetup/gpio_list exists it is used in place
	of the one in the package directory.
	This accommodates Relay HATs, etc. that use different GPIO pins.
```
How do I get this to be recognised - a reboot does not pick up the change. Do I have to clear a flag so it thinks it is a new version? 
Peter

kwindrem, 2022-10-30:
You need to run /data/RpiGpioSetup/setup again after adding the custom file.

pdcurtis, 2022-10-30:
@dersch81 @kwindrem Thanks. Checked with different GPIOs and does not seem to be problem with particular GPIOs so there does seem to be a limit in total number Victron Node recognises of 4. Worth mentioning in Victron Community Modification Space?

pdcurtis, 2022-11-06:
@dersch81 @kwindrem There is obviously another way to access Relays 5 and 6 and that is via the D-bus. That should integrate fully with RpiGpioSetup and GuiMods and also with Node-RED by using Exec Nodes containing 'Commands lines' 

There are at least two ways to access the D-bus under the Venus OS from the Command Line. The 'standard' low level way common to most machines is via `dbus-send` which actually Sets and Gets values. The other more focused solution is `dbus` which I understand is implemented using Python but is very slow even for the simplest tasks. The invocation using `dbus` is however much shorter and easier to understand. `dbus-spy` is a useful Victron utility which enable one to find out something about the D-bus configuration and current Values in a simple way. 

Lets have a look at how both look in a terminal (or Node-RED Exec node) noting the low level numbering for relays starts at 0. These are all single lines when you do a copy/paste 

### First using dbus-send 

**Turn Relay 5 on** 

`dbus-send --system --print-reply --dest=com.victronenergy.system /Relay/4/State com.victronenergy.BusItem.SetValue variant:int32:1`

**Turn Relay 5 off** 

`dbus-send --system --print-reply --dest=com.victronenergy.system /Relay/4/State com.victronenergy.BusItem.SetValue variant:int32:0`

**Get Value of Relay 5's State** 

`dbus-send --system --print-reply --dest=com.victronenergy.system /Relay/4/State com.victronenergy.BusItem.GetValue`

which returns the following 

```
method return time=1577563971.665778 sender=:1.33 -> destination=:1.912 serial=19248 reply_serial=2

variant int32 1
```

**Example if you want the command to only return the value of the switch by piping through awk:**

```
root@raspberrypi4:~# dbus-send --system --print-reply --dest=com.victronenergy.system /Relay/4/State com.victronenergy.BusItem.GetValue | awk -F'int32 ' '{printf "%s",$2}END{print '\n'}'
1
root@raspberrypi4:~#
```

### Now using the slow but simple call to dbus

**Turn Relay 6 on** 

`dbus -y com.victronenergy.system /Relay/5/State SetValue 1`

**Turn Relay 6 off** 

`dbus -y com.victronenergy.system /Relay/5/State SetValue 0`

**Example of Get Value of Relay 6's State** 

```
root@raspberrypi4:~# dbus -y com.victronenergy.system /Relay/5/State GetValue
1
root@raspberrypi4:~# 
```

Although the use of `dbus `looks much shorter and easier it is noticeably very slow ~0.5 sec versus 0.03secs for `dbus-send` on my Raspberry Pi 4B, a relatively fast machine, it can take nearly two seconds on some machines. This is probably due to the of python in the coding. If you want to check just put a `time` command at the very beginning of the command lines and try both on your machine. I will use `dbus-send` myself.

I have tested the `dbus-send` solution for both Setting and Getting Values in Node-RED Exec nodes. This gets round the problem of Relays 5 and 6 not being accessible from Victron Nodes

dersch81, 2023-03-20:
@pdcurtis that's the way. Thanks for your investigation :)

closed 2023-03-20 by dersch81