kwindrem/RpiGpioSetup #7: PullUp for GPIO 20/21

wolfganghuse, 2022-11-19:
Hello,
i am using your script with a Pimoroni AutomationHat... Relais Control is fine but that Hat uses GPIO 26/20/21 for three Inputs... Your Overlay only enables Pullup for 26, is it possible to add this also for the two other Pins ?

Regards,
Wolfgang

kwindrem, 2022-11-19:
The changes I made to pull-ups/downs was to support the digital inputs. When an GPIO is programmed for output as would be the case to drive relays, the pull-up/down is no longer a factor so this won't change anything for you.

Note that there is a period of time during boot where the GPIO pins retain their default settings. the overlay is applied some time during boot to reprogram the pull-ups/downs. But until that time the pins will be in their default state. This could cause a relay to close momentarily during boot. This is why I recommend active high relays for use with Venus OS.

Since you need to modify the gpio_list to remove/move digital input 5 from GPIO26 to something else, you should also be able to modify the overlay. I have provided the source file which needs to be compiled into a binary overlay. Instructions for this are in the ReadMe but you essentially need to do this under Raspberry PI OS since the overlay complier is not included in Venus OS.

wolfganghuse, 2022-11-19:
Thanks for the information... the Pull-Up request targets Input-Pins, there are no Relays at 20/21.
I will have a look at the Overlay-Details and create a customized overlay file.

closed 2022-11-19 by wolfganghuse