kwindrem/RpiGpioSetup #8: Temperature Relay needs battery temp option

lfhoward, 2023-01-21:
Hi Kevin, thanks for creating GUImods and its associated programs. I am using Venus OS on a Raspberry Pi and am excited to have 6 working relays. I want to use one of them to turn on battery heaters when the battery temperature falls below a certain level. In the temperature relay parameters I can select my ruuvitags (outdoor and indoor temp) but do not see the battery temperature from my BMV-712 battery monitor. Ideally this is what I need in order to reliably keep the battery above freezing. Since the battery monitor’s temperature reading is present on the dashboard and known to the Venus OS, would you be able to program the temperature relay to be able to use it? Thank you for considering this mod!

kwindrem, 2023-01-21:
The temperature relay logic is not something I've looked into. I had to do some special coding to extract the temperature from the battery monitor and suspect that isn't part of the relay logic. At this point I don't have plans to make any changed there.

lfhoward, 2023-01-21:
Thank you for your prompt reply. I do appreciate that. I will run some wiring to the physical relay on the back of the battery monitor as a workaround. Have a good weekend,Lauren