kwindrem/RpiGpioSetup #9: GPIO input with high 3,3 V

dahlei, 2023-01-27:
[image]

I want to use 4 digital inputs for my 4 wastewater non-contact sensors, will need to levelshift from 5 V to 3,3. The inputs from RPi are positive 3,3 V, they trigger in Venus OS when they are connected to GND. Is it possible to invert the GPIO ?

dahlei, 2023-01-27:
[image]

kwindrem, 2023-01-27:
Most configurations within Venus OS have an invert switch

dahlei, 2023-01-27:
I can invert from Open to Close in view, but GPIO to GND changes state. I want to use +3,3V to trigger change

kwindrem, 2023-01-27:
The GPIO inputs used by Venus are programmed to have an internal pull-up active, expecting a connection to ground from the external device. You can override this pull-up with a  smaller resistance externally.

You can also build a Device Tree Overlay that reprograms the inputs to have pull downs. I actually reprogram some of the pins that default to pull downs to pull ups so you can modify this as you like.

dahlei, 2023-01-27:
Yeah, I tried a 1K resistor and it is working. Thank you!