kwindrem/RpiGpioSetup #10: the relays disappear from the GUI

seamaster101, 2023-02-10:
I had it setup and working properly then one day after I moved the unit and powered it back up, there and no more relays shown in the GUI. I have waveshare hat with 3 relays and it was working perfectly well. The pins are setup properly in gpio_list and have no conflicts with DI 
I trued installing and reinstalling RPIGPIOSetup, but still mo relay on the GUI or on the settings
how do we get them back?

kwindrem, 2023-02-10:
RpiGpioSetup has no operating parts. It simply sets things up when running the setup script then backs out of the way.

Check /sys... for your relays and see if you can open and close the relays by writing 1 or 0 to that file. If not there may be something wrong with the hat.

seamaster101, 2023-02-10:
Thank you for the direction. I was successful of switching the all 3 relays on the hat with writing 1 in the files
`root@raspberrypi2:/sys/class/gpio# ls
export       gpio13       gpio16       gpio19       gpio20       gpio21       gpio26       gpio5        gpio6        gpiochip0    gpiochip504  unexport
root@raspberrypi2:/sys/class/gpio#  cd gpio21
root@raspberrypi2:/sys/class/gpio/gpio21# ls
active_low  device      direction   edge        power       subsystem   uevent      value
root@raspberrypi2:/sys/class/gpio/gpio21#  cd valua
-sh: cd: valua: No such file or directory
root@raspberrypi2:/sys/class/gpio/gpio21#  cd value
-sh: cd: value: Not a directory
root@raspberrypi2:/sys/class/gpio/gpio21#  nano value
root@raspberrypi2:/sys/class/gpio/gpio21# cd ..
root@raspberrypi2:/sys/class/gpio#  cd gpio20
root@raspberrypi2:/sys/class/gpio/gpio20#  nano value
root@raspberrypi2:/sys/class/gpio/gpio20# cd ..
root@raspberrypi2:/sys/class/gpio#  cd gpio26
root@raspberrypi2:/sys/class/gpio/gpio26#  nano value
root@raspberrypi2:/sys/class/gpio/gpio26# `
I still  dont see relays in the UI. What should I check next?

kwindrem, 2023-02-10:
Do you have the relays in the  Settings / Relay menu ?

If not, verify /etc/venus/gpio_list is correct. I think you said it was but ....

If so, do the on/off toggles close/open your relays?

Check Settings / Display & language / GuiMods / Show Relay overview is turned on. That will only affect whether you see the Relay Overview page.

seamaster101, 2023-02-10:
there is no RELAY listed in the SYSTEM menu

kwindrem, 2023-02-10:
It's in the Settings menu, not the System menu

seamaster101, 2023-02-10:
Sorry I misspoke… you are correct. I wad there multiple times. There was no relay listed as a menu item, but after  toggled the relays by writing 1 in the files as you advised the relay menu showed up now!
So did show the relay controls in the GUI. Very strange! Previously I restarted the device multiple time and checked. I was always getting grey page for the relays in GUI_mods and the relays was not listed in the settings menu
I don't know how to explain that… it works now but i hope it is not going to disappear again. You can close this issue now.
Thank you!

closed 2023-02-10 by kwindrem