kwindrem/RpiGpioSetup #13: Relay 1 not controllable from nodered

PappaD, 2023-02-27:
Hello and thank you for a great little kit for enabling relays.

I have relays up and running and control them from Remote Console, and all relays but "Relay 1" can be controlled by nodered. I thought that was a colission with the functions but even after setting the relay function to manual I cannot control "Relay 1" from nodered. Anyone knows why?

[image]

[image]

PappaD, 2023-02-27:
[image]

kwindrem, 2023-02-27:
This is not a function of any of my packages. Might ask in the Victron forum.
There's a special node red section of the forum