kwindrem/RpiGpioSetup #14: Only first two relays showing up on console

PappaD, 2023-02-27:
I have relays on a HAT and they show up in nodered (four of them according to the other issue).  In /dev/gpio I can see all six. But I can only see two in Remote Console:

[image]

[image]

[image]

Any idea how to make them visible in the console?

PappaD, 2023-02-27:
[image]

kwindrem, 2023-02-27:
You would need GuiMods to see the additional relays.