kwindrem/RpiGpioSetup #19: RpiGpioSetup V4.2

Gexle, 2023-12-09:
Hello Kevin

I updated RpiGpioSetup and after the update it said that it was no longer compatible with version V2.93...

Can you please fix this so that I can use RpiGpioSetup again please!

I hope it's not all gone now so my settings

Use RpiGipo for fan control

Gexle, 2023-12-09:
I made a backup beforehand.

If it works again with version V4.2, can I restore the backup or??

kwindrem, 2023-12-09:
You can set the branch/tag to v4.2 to get back to the older version. 

I find this odd since I've made no changes in the file sets for older versions. If you can get to the logs I'd very much like to see /data/log/SetupHelper where the errors are displayed. It will take me a while to load v2.93 on my test system.

kwindrem, 2023-12-09:
Forget my request for logs. I have identified the problem and working on a fix now.

kwindrem, 2023-12-09:
Fixed in RpiGpioSetup v4.3 and RpiDisplaySetup v3.7.

closed 2023-12-09 by kwindrem