kwindrem/RpiGpioSetup #20: How to adjust for relays closing low

tobelix87, 2024-01-07:
Hi,
I read the note that this extension is currently not working for relays which close low (connecting control cable to GND). However, I do have such relays and would like to use them. I will switch just fans with it and I am not rebooting my rpi too often. So it will not be a big issue if the fans turn on for a few seconds during boot process.

I assume that I need to adjust the Overlay to activate a pull up resistor for the GPIOs. But I am not very familiar with the GPIO setting, therefore it would be great if anyone could give me a hint.

Thanks, Tobelix

tobelix87, 2024-02-24:
I solved it in the following way:

1. created a new dtoverlay which configured the 6 output GPIOs for the relays with pull-ups. You can check the changes here: [https://github.com/kwindrem/RpiGpioSetup/pull/23](https://github.com/kwindrem/RpiGpioSetup/pull/23)
2. The switch will have inverted status. You can solve it as explained in this [issue](https://github.com/kwindrem/RpiGpioSetup/issues/21)

closed 2024-02-24 by tobelix87

kwindrem, 2024-02-24:
The issue is there is a period of time during boot before the DT overlay loads where relays that activate with a pull down will close. For this reason, I can not support this as a general fix. In your application, you don't care if the fan runs momentarily but if for example the relay is used to start a generator or other significant action then such a change will not be appropriate.

tobelix87, 2024-02-25:
that is understandable and totally makes sense. Nevertheless, just wanted to document it for other people who might have similar non-critical devices behind their relays