kwindrem/RpiGpioSetup #21: waveshare rpo Relay board - inverted status

antonstroberg, 2024-01-11:
I'm using rpi 3b+ with venus os 3.2 large install, have gui_mods and RpiGpioSetup installed and configured for this relay hat:
https://www.waveshare.com/wiki/RPi_Relay_Board
(waveshare with 3 relays). 

I have changed the /etc/venus/gpio_list so that my relays are connected to correct BMC vs channel and I can control all three relays via panels and in settings. So far so good. 
My problem is that status is inverted for all of them - on is off and off is on - so to say.  This becomes a bit of a problem since I have a temperature sensor and want to use the automatic temp settings for one of the relays. 

If this is specific to my relayboard - is there a way to invert_switch:true or similar :) 
kindly
Andreas

kwindrem, 2024-01-12:
Some of the relay boards need an active low signal. There are ways to invert the logic however at boot, the relay would still close until a device tree overlay can load. Without a DT overlay, the relay would remain closed until Venus OS was fully booted and the service controlling the relay gets a chance to configure the relay. For these reasons, I didn't consider adding a relay invert to be at all useful.

antonstroberg, 2024-01-12:
ok, I found a setting to control that under /dev/gpio/relay_1/active_low (for relay_1) - if I change the number in that file from 0 to 1 that relay act as expected. Problem is that once rebooted it will of course go back to 0 - Im not sure what script or config that sets these parameters?

antonstroberg, 2024-01-12:
found a solution that works - but it feels like Im switching off something that has been switched on earlier in boot. 
-------------
 I find a problem. The relay is inverted by default. Izak told me that I could try to change from 0 to 1 into /dev/gpio/relay_1/active_low, and it works, but until i reboot the board.

Then, something forces this to change from 1 to 0 when the board boots up.
I don't know for sure (haven't tested it) but I think this should work for you:

1. Create a file called /data/rcS.local with the contents:

echo 1 >/dev/gpio/relay_1/active_low

2. Make the file executable:

chmod 755 /data/rcS.local

kwindrem, 2024-01-12:
The problem is the chip itself defaults to an active high. When you reboot or turn on the PI, all the pins that default to outputs go low. The earliest opportunity to change a pin from active high to active low is in a device tree overlay which happens early on in the boot process but there is still several seconds of the "wrong" state.

To make this work without any glitches, you need to look for a relay board that works with an active high condition, or invert the signal before it gets to the board. If it is a hat, then that couldn't happen.

antonstroberg, 2024-01-12:
i solved it by adding:
echo 1 >/dev/gpio/relay_1/active_low
echo 1 >/dev/gpio/relay_2/active_low
echo 1 >/dev/gpio/relay_3/active_low

into /data/rcS.local 

not optimal but it seems to work after reboot.