kwindrem/RpiGpioSetup #22: Input 4 GPIO19

dersch81, 2024-02-21:
With the Waveshare 2CH FD in combination with RpiGpioSetup i notice a problem with GPIO19 even i don't use it due to alternate GPIO list. can1 of the can hat does not work because of this.

```
dmesg | grep spi
[    4.420357] pinctrl-bcm2835 fe200000.gpio: pin gpio19 already requested by fe200000.gpio; cannot claim for fe215080.spi
[    4.432072] pinctrl-bcm2835 fe200000.gpio: error -EINVAL: pin-19 (fe215080.spi)
[    4.457410] spi-bcm2835aux fe215080.spi: error -EINVAL: Error applying setting, reverse things back
[    4.651405] spi-bcm2835aux: probe of fe215080.spi failed with error -22
[    5.375888] spi_master spi0: will run message pump with realtime priority
[    5.434183] mcp251xfd spi0.0 can0: MCP2518FD rev0.0 (-RX_INT -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD c:40.00MHz m:20.00MHz r:17.00MHz e:16.66MHz) successfully initialized.
```

Since i'm using the valid gpio_listForCanHats, where GPIO19 is replaced with GPIO3 i'm wondering what is blocking the GPIO19 now. 

I found in VenusGpioOverlay.dts that GPIO19 is changed to pull-up via the overlay. So i have replaced 19 with 3, too. And reinstalled the RpiGpio that the overlay will match the new list. 

But the problem still persists. GPIO19 is blocked. 

I can't find any hint why. 

I also don't need any Input. Only the outputs to swich external relays are important for me. 

Is there any way to exclude the use of GPIO's for Inputs in general? How can i do that?

kwindrem, 2024-02-21:
You can remove all the input lines from gpio_list to free up the input pins.

By default, the Raspberry PI defines the first 8 GPIOs with pull-ups, the rest with pull-downs. That's the reason for VenusGpipOverlay.dts.

You need to look at what RpiGpio does to find the cause of the GPIO19 conflict. It's not an issue in VeCanSetup or RpiGpioSetup. RpiGpio is not my package.

dersch81, 2024-02-22:
The cause is the VenusGpioOverlay.dtbo what will inserted into the u-boot/config.txt by the setup of RpiGpioSetup. 
There is Pin19 set to high but also to IN. But the function need to be alt4 which will be set by the dtoverlay=spi1-3cs. 
That is causing the conflict. In my case it is easy to just remove VenusGpioOverlay because i don't use any input. 

But in case of all together a modified VenusGpioOverlay need to be compiled with dtc. But dtc is not available on VenusOS. 

Maybe a comment at the Readme can be placed. A better but with higher effort need a modified RpiGpioSetup with the option to choose another precompiled alternate dtbo file.

kwindrem, 2024-02-22:
Thanks for digging into this. RpiGpioSetup does offer an alternate gpio_list but I did not remember the overlay to switch to pull-ups. I'll work on an alternate overlay.

dersch81, 2024-02-22:
I have modifed the VenusGpioOverlay.dts in that way. But couldn't compile to .dtbo due to the missing dtc on VenusOS. 

It is not good for general use and was a try to fix my specific situation. But it could give you an idea what could work:

```
/dts-v1/;
/plugin/;

/ {
    compatible = "brcm,bcm2735,bcm2736,bcm2737,bcm2737A0,bcm2737B0,bcm2711";

// make all Venus GX digital inputs have pull-ups
// the chip default is GPIO 1 - 8 have pull-ups, the rest are pull-downs

// compile with
// dtc -@ -I dts -O dtb -o VenusGpioOverlay.dtb VenusGpioOverlay.dts

    fragment@0 {
        target = <&gpio>;
        __overlay__ {
            pinctrl-names = "default";
            pinctrl-0 = <&my_pins>;

            my_pins: my_pins {
                brcm,pins = <5 6 13 16 19 26>;     /* gpio no. */
                brcm,function = <0 0 0 0 4 0>; /* 0:in, 1:out */
                brcm,pull = <2 2 2 2 1 2>;     /* 2:up 1:down 0:none */
            };
        };
    };
};

```

But replacing the dtbo overlay file with the GPIO's of the alternate GPIO list will be more suitable, i guess.

kwindrem, 2024-02-22:
A new version of RpiGpioSetup (v4.4) should remove this issue. It installs a different overlay if the alternate gpio_list is selected.

closed 2024-02-22 by kwindrem

dersch81, 2024-02-22:
I'm not sure why and need to check more in detail but some new problem with shutdown GPIO16 now happens after the commit. GPIO16 is not used by the waveshare CAN FD HAT but the conflict is releated to it what never happend before. I'm not sure why at the moment.

```
[    4.433787] pinctrl-bcm2835 fe200000.gpio: pin gpio16 already requested by fe200000.gpio; cannot claim for fe215080.spi
[    4.444885] pinctrl-bcm2835 fe200000.gpio: error -EINVAL: pin-16 (fe215080.spi)
[    4.447745] uart-pl011 fe201000.serial: no DMA platform data
[    4.452362] pinctrl-bcm2835 fe200000.gpio: error -EINVAL: could not request pin 16 (gpio16) from group gpio16 on device pinctrl-bcm2711
[    4.470496] spi-bcm2835aux fe215080.spi: error -EINVAL: Error applying setting, reverse things back

```

dersch81, 2024-02-23:
It seems to be related to overlay spi1-3cs 

https://github.com/Ysurac/raspberry_kernel_mptcp/blob/master/arch/arm/boot/dts/overlays/spi1-3cs-overlay.dts

GPIO16 is in use there. I have no idea why all my manual testings never produced any conflict until now. I have tested spi1-1cs.dtbo now and this fixes the problem.

So we need another commit at the VeCanSetup Side and this should be the final fix in this

kwindrem, 2024-02-23:
Done. You will need to install RpiGpioSetup AND VeCanSetup.

I think the reason you haven't seen the GPIO 16 error previously is that the GPIO 19 error prevented checking for others.