kwindrem/RpiGpioSetup #24: RpiGpioSetup 4.5 -> Error no files set for V2.93

Gexle, 2024-02-28:
Hello Kevin 

Please fix the problem. Thank you

[image]

RpiGpioSetup 4.5 -> Error no files set for V2.93

Thanks

kwindrem, 2024-02-28:
RpiGpioSetup installs fine on my Rasberry PI and Venus OS v2.73.

My guess is that either you have a corrupted package (try downloading again) or that some other modification is conflicting with RpiGpioSetup. One known conflict is RemoteGPIO.

I would need to see the logs to investigate further. You can use the backup facility inside PackageManager to store the log files on USB stick then you can post the .zip file here.

Gexle, 2024-02-28:
Hello Kevin

I haven't installed anything else except your packages

kwindrem, 2024-02-28:
I'd recommend attempting to download the package again and see if will install then. You may need to reboot to clear the no file set error before you can attempt install.

Next I would recommend reinstalling the firmware but with an older version that requires downloading the appropriate .swu file from here:

http://updates.victronenergy.com/feeds/venus/

So you might consider upgrading the GX device's firmware to the current release from the Online updates menu.

Again, for me to help you troubleshoot further, I'd need the logs. I can not reproduce the issue here and that one screen shot only tells me that the install failed. The logs should tell me WHY it happened.

Gexle, 2024-02-28:
I've already done all of that, I'm currently looking for the logs, where are they in the Venus OS? Current??

Gexle, 2024-02-28:
[current.log](https://github.com/kwindrem/RpiGpioSetup/files/14437851/current.log)

kwindrem, 2024-02-28:
If you use PackageManager's backup function you have the choice of storing the logs.zip to a USB stick or to /data on the GX device.

The link you posted above does not seem to work. The one in your email is only for PackageManager and the one for SetupHelper will provide more detail.

I still recommend you reinstall the firmware if you have not done so already. v2.93 is available in the released section of feeds linked above.

Gexle, 2024-02-28:
I pulled the log from the USB

[logs.zip](https://github.com/kwindrem/RpiGpioSetup/files/14437953/logs.zip)

Gexle, 2024-02-28:
The rest of the Update Setuphelper Guimods Can has everything ok no problems

kwindrem, 2024-02-28:
Apologies. It seems RpiGpioSetup v4.5 on GitHub is corrupted. The corruption has to do with the program I use to update the package file sets: it leaves links to other file sets in place after the file set is removed!

The reason I didn't see it is my test system had a version I had not released yet so the install succeeded. Once I download the package from GitHub I could reproduce the failure.

I'm pushing v4.6 out now and this should fix the problem.

closed 2024-02-28 by kwindrem