kwindrem/RpiGpioSetup #25: When installing, 1W devices are gone

MBj1703, 2024-07-25:
Hi,

I have two 1w Tempsensors on GPIO 4.
When I install the RpiGpioSetup the devices are gone?

Any idea, what to change in my config?

Thanks!

MBj1703, 2024-07-25:
just figured out, when I delete _dtoverlay=VenusGpioOverlay_ out of _the /u-boot/config_ all is working as expected

kwindrem, 2024-07-25:
That overlay is changing the programming of some of the GPIOs so that all inputs have pull ups. The default for the chip is some to have pull downs. Those with pull downs are active when there's nothing connected.

So I suspect what you are seeing is a conflict between pins used by the 1 wire sensors with their use as digital inputs. You can customize the overlay and then provide a custom gpio_list to avoid those conflicts.

The overlay is not modified by a firmware update so you would only do this once. RpiGpioSetup has an option for a custom gpio_list and that would be reinstalled on a firmware update.

closed 2024-07-25 by kwindrem