kwindrem/RpiGpioSetup #26: CM4-IO-WIRELESS-BASE-B configure ports

Nvanderbeek, 2024-09-14:
I'm struggle-ing but willing to find out how to make a rpigpio setup for this board.
any way forward to map de UART RS485 and CAN port of this board.

I have found de pdf with pinout on you're side but struggle to match it with this board. Any clues?

kwindrem, 2024-09-14:
My VeCanSetup package may be able to help with the CANbus port but you may need to tweak some things. (There are often GPIO conflicts between devices.)

Others have found ways to activate an RS485 port on similar modules but I don't know anything about what they did or problems they encountered.

Nvanderbeek, 2024-09-15:
Hardware and ports are enabled. For Can and. Rs485. But the mapping to the
gpio pins avoid conflicts and add a relay is unclear for me at the moment