kwindrem/RpiGpioSetup #29: control from node-red

lehmpiotr, 2025-03-18:
is it possible to control relays from node-red?

kwindrem, 2025-03-18:
Others have done it but I have no experience with Node-Red. Sorry.

You might post in the Victoron community forum

closed 2025-03-18 by kwindrem

lehmpiotr, 2025-03-18:
The solution is prosaic :D

`[
    {
        "id": "11cbe7416f3c484f",
        "type": "victron-output-relay",
        "z": "b01f27ceeed75615",
        "service": "com.victronenergy.system/0",
        "path": "/Relay/2/State",
        "serviceObj": {
            "service": "com.victronenergy.system/0",
            "name": "Venus device"
        },
        "pathObj": {
            "path": "/Relay/2/State",
            "type": "enum",
            "name": "Venus relay 3 state",
            "enum": {
                "0": "Open",
                "1": "Closed"
            },
            "writable": true
        },
        "initial": "",
        "name": "",
        "onlyChanges": true,
        "x": 840,
        "y": 1160,
        "wires": []
    },
    {
        "id": "aa54b8193260ea50",
        "type": "victron-input-relay",
        "z": "b01f27ceeed75615",
        "service": "com.victronenergy.system/0",
        "path": "/Relay/2/State",
        "serviceObj": {
            "service": "com.victronenergy.system/0",
            "name": "Venus system"
        },
        "pathObj": {
            "path": "/Relay/2/State",
            "type": "enum",
            "name": "Venus relay 3 state",
            "enum": {
                "0": "Open",
                "1": "Closed"
            }
        },
        "initial": "",
        "name": "",
        "onlyChanges": true,
        "x": 220,
        "y": 1160,
        "wires": [
            [
                "36e56445fa45be12"
            ]
        ]
    },
    {
        "id": "1ffc4e312d5dede0",
        "type": "ui_switch",
        "z": "b01f27ceeed75615",
        "name": "",
        "label": "switch 3",
        "tooltip": "",
        "group": "102c64759ace5093",
        "order": 1,
        "width": "8",
        "height": "1",
        "passthru": true,
        "decouple": "false",
        "topic": "payload",
        "topicType": "msg",
        "style": "",
        "onvalue": "1",
        "onvalueType": "num",
        "onicon": "",
        "oncolor": "",
        "offvalue": "0",
        "offvalueType": "num",
        "officon": "",
        "offcolor": "",
        "animate": false,
        "className": "",
        "x": 600,
        "y": 1160,
        "wires": [
            [
                "11cbe7416f3c484f"
            ]
        ]
    },
    {
        "id": "36e56445fa45be12",
        "type": "debounce",
        "z": "b01f27ceeed75615",
        "time": "100",
        "name": "",
        "x": 460,
        "y": 1160,
        "wires": [
            [
                "1ffc4e312d5dede0"
            ]
        ]
    },
    {
        "id": "102c64759ace5093",
        "type": "ui_group",
        "name": "Relays",
        "tab": "2e029130b699bcc0",
        "order": 1,
        "disp": true,
        "width": "8",
        "collapse": true,
        "className": ""
    },
    {
        "id": "2e029130b699bcc0",
        "type": "ui_tab",
        "name": "Home",
        "icon": "dashboard",
        "disabled": false,
        "hidden": false
    }
]`